mp-se/victron-receiver #2: MQTT setup

thomasjobs, 2024-11-19:
I'm starting a new topic - MQTT. I'm not able to listen to the Victron Reader. 

All devices on a local network. Publish and listen to topics / packet works fine from HA. Also when enabling MQTT on Shelly device. When connecting with MQTT Explorer I have to turn off "Encryption (tls)" to make it connect. Then it's possible to publish and listen to topics in HA MQTT.

Serial console show "MQTT publish failed.....".

Attached my settings in Victron Receiver. Any advice?

[image]

[image]

mp-se, 2024-11-19:
The error -3 means that the receiever (MQTT Server) does not accept the data sent. It can connect but does not accept the data. In order to figure out why you need to look at the server log. 

This not a http protocol connection so you should only use the server name or IP, skip the http:// part. 

You used port 1883 which indicate that the server is not using SSL, then the port should be higher than 8000 as per the typical MQTT server settings. 

The software does support SSL connection but then the port needs to be higher than 8000 or else it will assume its non encrypted. 

I'm using MQTT for my HA integration on my local network so I know it works.

thomasjobs, 2024-11-19:
Brilliant! I changed my settings according your suggestions - now it works! Will continue to explore......

Thanks!!!

[image]

[image]