mp-se/victron-receiver #3: REST API setup

thomasjobs, 2024-11-21:
Have tried in vain to setup and get the REST API, for HA, to work. Have to my best knowledge followed the guide [here](https://community.home-assistant.io/t/use-nabu-casa-url-to-use-ha-api/382563), but not every syntax is understood by me (and hard to find explanations). So I seek assist here :-)

My setup is on a local network, HA have IP 192.168.1.145. My Victron Reader is also on my local network with IP 192.168.1.195. Have installed esphome and edited the config file. When connecting to the Serial Console I get error messages: HTTP post failed, response=-1

Anything obvious for you experts?

[image]
[esphome-web-9b6dcc.yaml.txt](https://github.com/user-attachments/files/17846815/esphome-web-9b6dcc.yaml.txt)
[Serial console.txt](https://github.com/user-attachments/files/17846816/Serial.console.txt)

mp-se, 2024-11-21:
The URL Should be the same base URL that you use to access Home Assistant. If you are the local network that should probably be http://192.168.1.145:8123/ if you are using SSL this is probably https://192.168.1.145:8123/

thomasjobs, 2024-11-21:
> The URL Should be the same base URL that you use to access Home Assistant. If you are the local network that should probably be http://192.168.1.145:8123/ if you are using SSL this is probably https://192.168.1.145:8123/

HTTP error code changed from -1 to 400 (when using http://192.168.1.145:8123/api/states/sensor.ChargerIP22/). If using https i get error code -1.

   1646273 I: PUSH: Sending values to HTTP post
   1646475 I: PUSH: Adding header 'Content-Type': 'application/json'
   1646677 I: PUSH: Adding header 'Authorization': 'Bearer Authorization: Bearer ****
   1647505 E: PUSH: HTTP post failed, response=400
   1647708 I: PUSH: Sending data to REST API for device ChargerIP22 and battery_voltage1
   1648012 I: PUSH: Sending values to HTTP post
   1648214 I: PUSH: Adding header 'Content-Type': 'application/json'
   1648416 I: PUSH: Adding header 'Authorization': 'Bearer Authorization: Bearer ****
   1649304 E: PUSH: HTTP post failed, response=400
   1649507 I: PUSH: Sending data to REST API for device ChargerIP22 and battery_current1

mp-se, 2024-11-21:
You should not add the api/... that will cause the error. Add the URL as I described above

mp-se, 2024-11-21:
Have created a long lived token in HA and added that to victron receiver ?

thomasjobs, 2024-11-21:
> You should not add the api/... that will cause errors since that part is added in the victron receiver when updating/adding sensor . Add the URL as I described above

Changed and now "response=-1"
   2244417 I: PUSH: Sending data to REST API for device ChargerIP22 and error_message
   2244720 I: PUSH: Sending values to HTTP post
   2244922 I: PUSH: HTTP, SSL enabled without validation.
   2245124 I: PUSH: Adding header 'Content-Type': 'application/json'
   2245326 I: PUSH: Adding header 'Authorization': 'Bearer Authorization: Bearer ****
   2246158 E: PUSH: HTTP post failed, response=-1
   2246360 I: PUSH: Sending data to REST API for device ChargerIP22 and battery_voltage1
   2246663 I: PUSH: Sending values to HTTP post
   2246865 I: PUSH: HTTP, SSL enabled without validation.
   2247067 I: PUSH: Adding header 'Content-Type': 'application/json'
   2247269 I: PUSH: Adding header 'Authorization': 'Bearer Authorization: Bearer ***
   2248104 E: PUSH: HTTP post failed, response=-1
   2248306 I: PUSH: Sending data to REST API for device ChargerIP22 and battery_current1
   2248609 I: PUSH: Sending values to HTTP post
   2248811 I: PUSH: HTTP, SSL enabled without validation.
   2249013 I: PUSH: Adding header 'Content-Type': 'application/json'
   2249215 I: PUSH: Adding header 'Authorization': 'Bearer Authorization: Bearer

mp-se, 2024-11-21:
-1 means connection refused so the server is not accepting the connection. You dont have a firewall that blocks it ? Or is the port number / server adress incorrect?

mp-se, 2024-11-21:
You should be able to enter the same URL into your browser and land on the home assistant login page (or dashboard)

mp-se, 2024-11-21:
If they are on the same network it should work fine. I dont know if the REST API needs to be activated in HA or not. I didnt have to do anything on my installation. There is not that much information on the topic; https://www.home-assistant.io/integrations/api/

mp-se, 2024-11-21:
If you point your browser to http://192.168.1.145:8123/api/states you should get error 401 due to the lack of authorization token

thomasjobs, 2024-11-21:
Will study curl command and thoroughly read this article...... https://funprojects.blog/2020/12/12/home-assistant-rest-api/

mp-se, 2024-11-22:
I use the POST /api/states in the victron receiver to add/update a sensor, documentation can be found here https://developers.home-assistant.io/docs/api/rest/

thomasjobs, 2024-11-22:
It seems REST API setup in HA is wrong........can not POST (simulate) a sensor in my HA.....I receive error codes.
So far have used CMD in windows10 but will switch to a Raspberry pi to rule out any interference from W10.

thomasjobs, 2024-11-22:
Some progress.....

I added http: to my configuration.yaml. Created new long lived token and double+tribble checked long lived token in Victron Receiver and esphome secrets.yaml. Now serial console is running without errors......next step is to create sensors in HA.

   5722790 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5724567 I: BLE : Starting PASSIVE scan.
   5725267 W: BLE : The stored encryption key does not match the device 3F vs 74
   5725440 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5728576 I: BLE : Starting PASSIVE scan.
   5729842 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5732578 I: Main: Name=ChargerIP22
   5732579 I: PUSH: Sending data to REST API for device ChargerIP22 and name
   5732580 I: PUSH: Sending values to HTTP post
   5732580 I: PUSH: Adding header 'Content-Type': 'application/json'
   5732581 I: PUSH: Adding header 'Authorization': '****
   5732602 I: PUSH: HTTP post successful, response=200
   5732665 I: PUSH: Sending data to REST API for device ChargerIP22 and state
   5732665 I: PUSH: Sending values to HTTP post
   5732666 I: PUSH: Adding header 'Content-Type': 'application/json'
   5732666 I: PUSH: Adding header 'Authorization': 'Bearer ****
   5732690 I: PUSH: HTTP post successful, response=200
   5732692 I: PUSH: Sending data to REST API for device ChargerIP22 and state_******   
   5732788 I: PUSH: HTTP post successful   
   5732839 I: PUSH: HTTP post successful    
   5732989 I: PUSH: HTTP post successful, response=200
   5733459 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=na   
   5735881 W: BLE : The stored encryption key does not match the device 3F vs 74
   5737227 I: BLE : Starting PASSIVE scan.
   5737331 W: BLE : The stored encryption key does not match the device 3F vs 74
   5737524 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5741243 I: BLE : Starting PASSIVE scan.
   5742146 W: BLE : The stored encryption key does not match the device 3F vs 74
   5742240 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5745252 I: BLE : Starting PASSIVE scan.
   5746329 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5746640 W: BLE : The stored encryption key does not match the device 3F vs 74
   5749271 I: BLE : Starting PASSIVE scan.
   5750448 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5750554 W: BLE : The stored encryption key does not match the device 3F vs 75
   5753280 I: BLE : Starting PASSIVE scan.
   5753298 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5753306 W: BLE : The stored encryption key does not match the device 3F vs 75
   5757399 I: BLE : Starting PASSIVE scan.
   5757512 W: BLE : The stored encryption key does not match the device 3F vs 75
   5757711 I: VIC : Victron AC Charger (A331) battVolt1=13.15 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5761409 I: BLE : Starting PASSIVE scan.
   5761622 W: BLE : The stored encryption key does not match the device 3F vs 75
   5762098 I: VIC : Victron AC Charger (A331) battVolt1=13.16 V battCurr1=0.00 battVolt2=nan V battCurr2=nan battVolt3=nan V battCurr3=nan temp=nan acCurrent=nan
   5765411 I: Main: Name=ChargerIP22
   5765413 I: PUSH: Sending data to REST API for device ChargerIP22 and name
   5765413 I: PUSH: Sending values to HTTP post
   5765413 I: PUSH: Adding header 'Content-Type': 'application/json'
   5765414 I: PUSH: Adding header 'Authorization': 'Bearer *****
   5765439 I: PUSH: HTTP post successful, response=200
   5765441 I: PUSH: Sending data to REST API for device ChargerIP22 and state
   5765442 I: PUSH: Sending values to HTTP post
   5765442 I: PUSH: Adding header 'Content-Type': 'application/json'
   5765442 I: PUSH: Adding header 'Authorization': '*****
Socket closed

(c) 2024 Magnus Persson

thomasjobs, 2024-11-24:
From my HA, after adding Victron sensors (230VAC is a Tuya sensor).
[image]

thomasjobs, 2024-12-13:
Based on my notes, during my work with Victron Reader, I have created a setup guide. 

@mp-se, there is a lot of interest in this project, but I think many potential users skip as they se it as to complicated. Maybe the guide could inspire some to try it out :-)

What is your thoughts?

mp-se, 2024-12-13:
Sure, any help is welcome. I have too many projects ongoing some things that might benefit this one as well. 

I dont know what your experience is but the easiest way to add documentation is to use the MD syntax as I do in the readme.md file. There are good guides on how that works.

Create a fork on the project and then you can edit in your copy and submit a PR that I can integrate. Just make a new file and add your docs to that. That way you will be listed as a contributor.

thomasjobs, 2024-12-13:
> Sure, any help is welcome. I have too many projects ongoing some things that might benefit this one as well.
>
> I dont know what your experience is but the easiest way to add documentation is to use [...]

Ok, will follow your advice. Have never done this…….untested area for me.

mp-se, 2024-12-15:
> Ok, will follow your advice. Have never done this…….untested area for me.

Give it a try, its not that difficult especially if you use Markup syntax, most project use that in some way so there are multiple examples. One good one is; https://github.com/bblanchon/ArduinoJson that is quite comprehensive.. 

I'm also testing a new flashing utility to make that much easier to handle. You can check it out here, https://mp-se.github.io/victron-receiver/flash/index.html

This will to a complete erase so be careful... Intended for first time flashing, upgrades should go through the device update feature. It's also possible to add WIFI settings using this method so this is the next thing to adress..

thomasjobs, 2024-12-16:
Example of setup in Home Assistant for the Victron Reader.

[image]

mp-se, 2024-12-16:
Closing this since i think it has been fixed

closed 2024-12-16 by mp-se