mp-se/victron-receiver #4: IP22 Charger - incorrect reading for "AC Current"

thomasjobs, 2024-12-12:
I discovered an incorrect value sent by the IP22 Charger - AC Current (ac_current). In my installation it's reported as 510 (A), correct value is ca 0,21A.

In the pdf ["Extra Manufacturer Data"](https://communityarchive.victronenergy.com/storage/attachments/extra-manufacturer-data-2022-12-14.pdf) IP22 Charger has a record for AC current. It seems as the Victron Reader is decoding this value incorrect. Or have I misunderstood?

[image]

[image]

[image]

mp-se, 2024-12-12:
The decoding is probably incorrect, i dont have any test data to check against

thomasjobs, 2024-12-12:
Hmmmmm…….what data do you need?

thomasjobs, 2024-12-12:
> Preferably a text version of the first image so i dont need to type it in and what the decoded values are in the victron app. Same as before

Ok, I fix.

thomasjobs, 2024-12-15:
Here the data from the reader.

[image]
[Data 15dec.txt](https://github.com/user-attachments/files/18140305/Data.15dec.txt)

mp-se, 2024-12-15:
Thanks, that was the data i needed, will release an update

closed 2024-12-15 by mp-se