mp-se/victron-receiver #9: WiFi setup

thomasjobs, 2024-12-23:
Upgraded firmware to 0.2.5 today. Unfortunately it's no longer possible to connect to a WiFi network. Have tried both in motorhome and at home. Result: Failed to get wifi scan status.

[image]
[image]

mp-se, 2024-12-23:
Where did you download the version from ? the Release page or directly from the dev stream ? The dev stream is work in progress and not full tested. 

Did you do the update via the upload firmware option in the UI ? It should retain all the configuration settings so everything should work as before? 

I've been running the same version for 2 weeks without any issues. 

In order to troubleshoot this you need to get the serial log from the device via serial cable and some software that can read that data. Putty is one of those that can be used.

thomasjobs, 2024-12-23:
ok, will troubleshoot…….

thomasjobs, 2024-12-23:
Downloaded the latest version and flashed the device again........now everything work fine.......strange....

After the weekend I will continue checking the data from the the IP22Charger.

[image]

mp-se, 2024-12-23:
Could have been some issue with previous flashing that can have strange effects

closed 2024-12-25 by mp-se