mp-se/victron-receiver #10: add support for esp32c3

WilliThomas, 2025-01-01:
Hi, can you add support for esp32c3?

mp-se, 2025-01-01:
Sure, what board do you have ?

WilliThomas, 2025-01-01:
The devkit without tft.

https://sigmdel.ca/michel/ha/esp8266/super_mini_esp32c3_en.html

mp-se, 2025-01-01:
Dont think that board is offically supported by arduino 2.x that I'm using but I can try to make a build for a generic c3 board which  probably works since I dont use any GPIO ports anyway

WilliThomas, 2025-01-01:
I can do the testing, if it works it works, if not we'll know something else.

mp-se, 2025-01-01:
You can try the version in the master branch, you can flash using the flash.py command in the root directory. If that works I will add it to the web flasher.

WilliThomas, 2025-01-01:
I think my esp32c3 has not enough memory: 
WARNING: Set --flash_size 16MB is larger than the available flash size of 4MB

serial monitor: 
21:51:59.918 -> ELF file SHA256: 6b695d78355b36ae
21:51:59.918 -> 
21:51:59.918 -> E (650) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
21:51:59.951 -> E (658) esp_core_dump_elf: Elf write init failed!
21:51:59.951 -> E (663) esp_core_dump_common: Core dump write failed with error=-1
21:51:59.951 -> Rebooting...
21:51:59.951 -> ESP-ROM:esp32c3-api1-20210207
21:51:59.951 -> Build:Feb  7 2021
21:51:59.951 -> rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
21:51:59.951 -> Saved PC:0x40048b82
21:51:59.951 -> SPIWP:0xee
21:51:59.951 -> mode:DIO, clock div:1
21:51:59.951 -> load:0x3fcd5820,len:0x1058
21:51:59.951 -> load:0x403cc710,len:0x8a8
21:51:59.983 -> load:0x403ce710,len:0x2fac
21:51:59.983 -> entry 0x403cc710
21:52:00.246 -> E (291) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.
21:52:00.246 -> 
21:52:00.246 -> assert failed: do_core_init startup.c:350 (flash_ret == ESP_OK)
21:52:00.246 -> Core  0 register dump:
21:52:00.246 -> MEPC    : 0x403819bc  RA      : 0x4038aa20  SP      : 0x3fcde140  GP      : 0x3fc92c00  
21:52:00.279 -> TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
21:52:00.279 -> S0/FP   : 0x00000001  S1      : 0x3fcde2af  A0      : 0x3fcde1a4  A1      : 0x3fc94c2d  
21:52:00.279 -> A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc9d000  
21:52:00.279 -> A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000088  S3      : 0x3fcde198  
21:52:00.312 -> S4      : 0x3fcde198  S5      : 0x420426f8  S6      : 0xffff0000  S7      : 0x3c164378  
21:52:00.312 -> S8      : 0x00090020  S9      : 0x00010020  S10     : 0x00000005  S11     : 0x3fcde3c0  
21:52:00.312 -> T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
21:52:00.312 -> MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
21:52:00.345 -> MHARTID : 0x00000000

WilliThomas, 2025-01-01:
Changed the 16MB to 4MB in flash.py. It seems to work now

mp-se, 2025-01-01:
Ok, change in the python file from 16mb to 4mb that should work fine

mp-se, 2025-01-01:
Good, i will add it to the webflasher tomorrow then

mp-se, 2025-01-02:
Updated webflasher with s3 and c3 options (no tft). closing this as completed.

closed 2025-01-02 by mp-se