mp-se/victron-receiver #12: Solar Voltage for solar charger

and117, 2025-01-25:
Is it possible to also get Solar voltage from solar charger? The voltage that is shown  is batteri voltage. I use that in my automations to disconnect the ac charger when there is enough sun.

[image]

mp-se, 2025-01-25:
Should not be a problem, let me have a look at it

mp-se, 2025-01-25:
I have checked the specification and those values are not included in the data from the solar charger profile, are there any other devices connected ? Temperature is also not part of the data frame.

[image]

and117, 2025-01-25:
I got the values on the Solar charger both from the Victron app and from a Vitron VRM that i run on a PI.

mp-se, 2025-01-25:
Ok, ill check the data packet if there is data not shown in the specification

mp-se, 2025-01-25:
There seams to be more data in the bluetooth packet than in the specification but it will probably take some time to figure out what it is and how to decode them.

mp-se, 2025-03-13:
Closing this due to inactivity and lack of needed input

closed 2025-03-13 by mp-se