mp-se/victron-receiver #13: Charger IP22 (using 3 outputs)

thomasjobs, 2025-02-02:
I'm using a charger IP22 12/30A with 3 outputs. I have recently changed my installation and connected the second and third output:
- First output is connected to my motor home battery.
- Second is used for 230VAC sense (current only a few mA).
- Third is used to charge my starter battery.

After installation I expected to se data from each output (volt & current) on the Victron Reader, currently Victron Receiver show only data for "battery_current1" and "battery_voltage1" (a SUM of all 3 outputs?). I'm lacking 2 and 3. According to specification the data should be available, ref ["Extra Manufacturer Data"](https://communityarchive.victronenergy.com/storage/attachments/extra-manufacturer-data-2022-12-14.pdf).

Magnus, appreciate if you have a look at it :-)

[Charger IP22_copy data.txt](https://github.com/user-attachments/files/18632383/Charger.IP22_copy.data.txt)
[header.txt](https://github.com/user-attachments/files/18632384/header.txt)
[raw.json](https://github.com/user-attachments/files/18632381/raw.json)
[image]
[Serial console.txt](https://github.com/user-attachments/files/18632380/Serial.console.txt)
[status.json](https://github.com/user-attachments/files/18632382/status.json)

mp-se, 2025-02-03:
This is the data that I get from the package, so the values for batt2 and batt3 are undefined - N/A, (0x1fff) thats why they are not shown in the receiver. So the charger is not sending the values for the other channels. 

State 6
Error 0
Batt1 (V) 13.76 0x560
Batt1 (A) 4.5 0x2d
Batt2 (V) 81.91 0x1fff
Batt2 (A) 204.7 0x7ff
Batt3 (V) 81.91 0x1fff
Batt3 (A) 204.7 0x7ff
Temp -2
Total (A) 51.1

thomasjobs, 2025-02-03:
Ok, thanks for feedback.

closed 2025-02-09 by mp-se