mp-se/victron-receiver #15: Feature request

thomasjobs, 2025-03-01:
Is it possible to add more information in the data transmitted?

- Minimum resend time
- Installed firmware

This information does not need to be sent frequently, once every 24h is enough.

Thanks!

thomasjobs, 2025-03-01:
And a possibility to set a user and password would be much appreciated :-)

mp-se, 2025-03-10:
> Is it possible to add more information in the data transmitted?
> 
> * Minimum resend time
> * Installed firmware
> 
> This information does not need to be sent frequently, once every 24h is enough.
> 
> Thanks!

I could add a sensor with that information.

mp-se, 2025-03-10:
> And a possibility to set a user and password would be much appreciated :-)

Set user and password for what ?

thomasjobs, 2025-03-10:
> > And a possibility to set a user and password would be much appreciated :-)
> 
> Set user and password for what ?

Today the device ip address is open, no protection at all. Just a bit concerned of the security………

mp-se, 2025-03-10:
> > > And a possibility to set a user and password would be much appreciated :-)
> > 
> > 
> > Set user and password for what ?
> 
> Today the device ip address is open, no protection at all. Just a bit concerned of the security………

Well, adding username and password would not really help with security since there is no encryption between the browser and the server so any passwords would be sent in clear text and not really secure... The arduino framework has no good support for that which would apply to most devices built with the arduino framework. 

My recommendation would be to secure the devices on a separate network from your normal devices and ensure that devices on that network cannot communicate with each other with a strict firewall.

thomasjobs, 2025-03-10:
Ok, thanks for clarifying. Not important.Sent from my iPhone. Brgds /Thomas+4670690588110 mars

mp-se, 2025-03-13:
You can try the dev build here and see if that works for the first topic; https://github.com/mp-se/victron-receiver/tree/dev/bin

closed 2025-03-19 by mp-se