mp-se/victron-receiver #16: Support Smart BatteryProtect

cschipper1, 2025-03-12:
Support Smart BatteryProtect see data below:

{"name":"Unknown","model":"0xA3B0","decrypted_data":"0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x74,0x0A,0x0C,0x00,0x08,0x00,0x00,0x00,0xFF,0x31,0x43,0x00,0x00,0x00","vendor_id":"0x02E1","beacon_type":"0x10","victron_record_type":"0x09"}

If you need more data or anything else let me know.

Thanks.

mp-se, 2025-03-12:
Would be good to have a screenshot of the values from the victron app to check if the decoding works

mp-se, 2025-03-13:
This is the data I could decode from that package, can you validate if those are correct ? Would also like some input to what the warning message was.

Victron Battery Protect (A3B0) state=0 error=0 out-state=0 alarm=0 warning=8 inputVolt=26.76 V outputVolt=0.12 VoffReason=10

Off reason should be; Remote input. + Protection active

cschipper1, 2025-03-13:
Hi,

The 8 is not a warning, it is the operation mode i configured:
https://www.victronenergy.com/media/pg/Smart_BatteryProtect_12V_24V/en/operation-and-programming.html#UUID-54509c0a-c857-eaf5-8278-4e2b834f13ca

mp-se, 2025-03-13:
That makes sense, then the data format document is a little confusing.. I will have a look and see how to present that part

mp-se, 2025-03-13:
you can try the version in the dev branch; https://github.com/mp-se/victron-receiver/tree/dev/bin

cschipper1, 2025-03-13:
I downloaded the
https://github.com/mp-se/victron-receiver/blob/dev/bin/firmware32c3.bin and
updated it via the webinterface, restarted but it does not recorgnized the
battery protect yet.

cschipper1, 2025-03-13:
Not all data is correct if turned on:

[image]

[image]

data with switch on:

{"name":"Battery Protect","model":"0xA3B0","decrypted_data":"0xF9,0x01,0x00,0x00,0x00,0x00,0x00,0x24,0x0A,0x23,0x0A,0x00,0x00,0x00,0x00,0x00,0x89,0x9E,0x00,0x00,0x00","state":249,"state_message":"Unknown","output_state":1,"error":0,"error_message":"0 - No error","input_voltage":25.96,"output_voltage":25.95,"alarm_reason":0,"warning_reason":0,"off_reason":30,"off_reason_message":"Switched off switch. Switched off register. Remote input. Protection active. "}

data switch off:

{
    "name": "Battery Protect",
    "model": "0xA3B0",
    "decrypted_data": "0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x6A,0x0A,0x02,0x00,0x08,0x00,0x00,0x00,0x00,0x22,0x9F,0x00,0x00,0x00",
    "state": 0,
    "state_message": "Off",
    "output_state": 0,
    "error": 0,
    "error_message": "0 - No error",
    "input_voltage": 26.66,
    "output_voltage": 0.02,
    "alarm_reason": 0,
    "warning_reason": 8,
    "off_reason": 30,
    "off_reason_message": "Switched off switch. Switched off register. Remote input. Protection active. "
}

mp-se, 2025-03-13:
Thanks, the documentation is not that clear on how to interpret the values. Looks like the output_state defines if the output is on or off. Im not sure how the offReason or state should be interpreted in those cases... Need to see if I can find some other documentation on how to interpret the data

mp-se, 2025-03-14:
I've done some changes to the decoding and found some errors. see if the new dev build works better

cschipper1, 2025-03-14:
Yes it is better, thanks! 1 on 2 off:

[image]

[image]

mp-se, 2025-03-14:
Perfect, I will fix the model string and do a release once I get confirmation on the other change I did