mp-se/victron-receiver #18: Increase security by adding SSL + Authentication options for web interface.

mp-se, 2025-03-16:
> > And a possibility to set a user and password would be much appreciated :-)
> 
> Set user and password for what ?
> 
> Today the device ip address is open, no protection at all. Just a bit concerned of the security……… 

 _Originally posted by @thomasjobs in [#15](https://github.com/mp-se/victron-receiver/issues/15#issuecomment-2709952748)_

This library might be an option for adding SSL support and increase security. https://github.com/hoeken/PsychicHttp

mp-se, 2025-03-19:
Got some progress on this topic, this should be good to implement in future releases of the project.

mp-se, 2025-03-23:
There is a beta build in the branch called ssl that I'm running at home now if you want to test it out. https://github.com/mp-se/victron-receiver/tree/ssl/bin

thomasjobs, 2025-03-23:
> There is a beta build in the branch called ssl that I'm running at home now if you want to test it out. https://github.com/mp-se/victron-receiver/tree/ssl/bin

Super! Will try in a day or two. Thanks!

mp-se, 2025-03-23:
The ui is quite slow do when ssl is enabled

thomasjobs, 2025-04-02:
> > There is a beta build in the branch called ssl that I'm running at home now if you want to test it out. https://github.com/mp-se/victron-receiver/tree/ssl/bin
> 
> Super! Will try in a day or two. Thanks!

Beta installed and work perfect! Esp32 display is blank?

[image]

mp-se, 2025-04-02:
What hardware do you have?

thomasjobs, 2025-04-02:
> What hardware do you have?

[image]

mp-se, 2025-04-02:
Doesn't really help, need to know the manufacturer

thomasjobs, 2025-04-02:
> Doesn't really help, need to know the manufacturer

[image]

mp-se, 2025-04-03:
the correct firmware for that board is; firmware32s3w_tft.bin

Here is a list of firmware vs hardware setup.

* Waveshare ESP32 S3 with TFT (firmware32s3w_tft.bin)
* Lolin ESP32 PRO with Lolin TFT (firmware32_tft.bin)
* Lolin ESP32 S3 PRO with Lolin TFT (firmware32s3_tft.bin)
* Lolin EP32 C3 without TFT (firmware32c3.bin)
* Lolin EP32 S3 without TFT (firmware32s3.bin)

thomasjobs, 2025-04-04:
Flashed the firmware today and all works fine (and the display ).

One detail is wrong,  “receiver app build”. At the moment is showing a hex value…….

[image]

mp-se, 2025-04-04:
Thats correct, its the github commit hash

mp-se, 2025-04-26:
Did you have a chance to test the ssl variant ?

thomasjobs, 2025-04-27:
> Did you have a chance to test the ssl variant ?

No, not yet.