mp-se/victron-receiver #19: Bad value on MPPT75 | 15 Smart Solar

Kogitae, 2025-03-17:
Hi good project, i use it with the MPPT75 | 15 smart solar
value are fine, but not in off state ( when no sun ) 

if Victron_state_message is Off
Victron_battery_current spike to 3276,6 A ( YEEEEEEE AAHHH )

glitch value for sure

mp-se, 2025-03-17:
Can you share the data packet when  there is a glitch?

Kogitae, 2025-03-17:
i will check under my mqtt client if i can find it this evening

mp-se, 2025-03-17:
I need the raw data packet from the web interface

Kogitae, 2025-03-21:
update to 0.5 done, 

good data : 

Victron
model string: Unknown device 0xA075
name: Solar Charger
model: 0xA075
decrypted_data: 0x03,0x00,0x1F,0x05,0x02,0x00,0x03,0x00,0x08,0x00,0x05,0xFE,0xFF,0xFF,0xFF,0xFF,0x07,0x2F,0x00,0x00,0x00
state: 3
state_message: Bulk
error: 0
error_message: 0 - No error
battery_voltage: 13.11
battery_current: 0.2
pv_power: 8
load_current: 0.5

i wait for bad one this evening

Kogitae, 2025-03-21:
no solar today so value are available 
Victron
model string: Unknown device 0xA075
name: Solar Charger
model: 0xA075
decrypted_data: 0x03,0x00,0xF3,0x04,0xFE,0xFF,0x04,0x00,0x02,0x00,0x04,0xFE,0xFF,0xFF,0xFF,0xFF,0x5E,0x56,0x00,0x00,0x00
state: 3
state_message: Bulk
error: 0
error_message: 0 - No error
battery_voltage: 12.67
battery_current: 3276.6
pv_power: 2
load_current: 0.4

does it help you ?

mp-se, 2025-03-21:
Thanks, that will help, what is the expected values?

Kogitae, 2025-03-21:
at night or when no charging Victron_battery_current spike to 3276,6 A normaly it's 0 not ?

mp-se, 2025-03-22:
Big thanks, this data was really helpful. The issue is that the value is negative current so it was not decoded properly, should be -0.2A. This was the first data I got that contained negative values so all my decoding of those was off. Will issue a new version

Kogitae, 2025-03-22:
fine good news, now that i can access and update this remote install without problem, i can test as many time you need

mp-se, 2025-03-22:
You can see if the build in the dev branch works properly; https://github.com/mp-se/victron-receiver/tree/dev/bin

Kogitae, 2025-03-22:
flash made, i wait for a no solar time for confirming

Kogitae, 2025-03-22:
by now change nothing
{"name":"Solar Charger","model":"0xA075","decrypted_data":"0x00,0x00,0x01,0x05,0xFC,0xFF,0x0B,0x00,0x00,0x00,0x04,0xFE,0xFF,0xFF,0xFF,0xFF,0x18,0xA6,0x00,0x00,0x00","state":0,"state_message":"Off","error":0,"error_message":"0 - No error","battery_voltage":12.81,"battery_current":3276.4,"pv_power":0,"load_current":0.4}

but i have doubt if the update go to the end ( same serial than the 0.5 )

Kogitae, 2025-03-22:
ok i understood, during the upload i look at the network process and app.js send me a error 500

Kogitae, 2025-03-22:
ok on my way, bad size donwload, and so bad CRC, i just redone with a new download attempt, update to 0.51 done 

here is the data : 

{"name":"Solar Charger","model":"0xA075","decrypted_data":"0x00,0x00,0x00,0x05,0xFD,0xFF,0x0B,0x00,0x00,0x00,0x04,0xFE,0xFF,0xFF,0xFF,0xFF,0xAB,0xA9,0x00,0x00,0x00","state":0,"state_message":"Off","error":0,"error_message":"0 - No error","battery_voltage":12.8,"battery_current":-0.3,"pv_power":0,"load_current":0.4}

so fine and the minus valus is for me due to the need on my setup of a scotchy diode to avoid back current to the panel setup . On your monitoring part, it seems fine, good job

mp-se, 2025-03-23:
Thanks for testing, I have published that release so closing this one

closed 2025-03-23 by mp-se