mp-se/victron-receiver #21: Unit problem for pv_power

Kogitae, 2025-09-10:
Hi with home assistant and the mqtt autodiscovery thhe mqtt message seems to have a problem 
In my case 

{"device_class":"current","unit_of_measurement":"W","unique_id":"D5209B967034_pv_power","device":{"identifiers":"D5209B967034","name":"Victron"},"name":"Victron_pv_power","state_topic":"victron_instant/D5209B967034/pv_power"}

will need to be 
{"device_class":"power","unit_of_measurement":"W","unique_id":"D5209B967034_pv_power","device":{"identifiers":"D5209B967034","name":"Victron"},"name":"Victron_pv_power","state_topic":"victron_instant/D5209B967034/pv_power"}

as 

https://www.home-assistant.io/integrations/sensor/#device-class

seems due to a recent change on home assistant

mp-se, 2025-09-10:
You need to do a full flash to use the 0.6 version since I have changed the flash layout in 0.6. So I recommend to do a backup of the config and then flash via usb

mp-se, 2025-09-10:
The 0.6 also allows you to enable SSL and have a secure / password protected remote access to the device.

closed 2025-10-19 by mp-se

Kogitae, 2025-10-21:
Hello, i find the time for taking back home my Esp32, and try to flash but i don't show any ESP32-C3 version for the 0.6 firmware . So hard to try